bmd-studio/stm32-for-vscode #213: Hiden icon？

chushangcs, 2024-10-07:
Hello, today when I opened the cubemx project with vscode, the icon of this plugin did not appear, and I did not hide the icon, but the icon appeared again after I re-opened the project several times. I would like to know whether the detection of the plug-in is not good enough, resulting in the timely detection of the cubemx project? Thank you!

jortbmd, 2024-10-07:
Hi. Thanks for opening up an issue. The icon should show itself once if there is an .ioc file present or the STM32-for-VSCode.config.yaml file. It might take a while for the icon to load based upon other extensions you have installed. I don't know if it shows up after a while? Another option is to run one of the available commands from the extension to at least show the icon.

jortbmd, 2024-12-23:
Thanks for opening up this issue. As there hasn't been any more information for a few months I will close the issue. Should it occur again or if you want to add more information to this issue do not hesitate te re-open the issue again.

closed 2024-12-23 by jortbmd