bmd-studio/stm32-for-vscode #214: When FATFS middle ware is added in CubeMX, clean rebuild fails.

devinw1, 2024-10-14:
I have tested this in a new project with STM32G431RBT6: Make a new project with makefile and generate code. Then go to VSCode and build and clean build work fine. 

Then go back to Cube MX and select to add FATFS middleware and regenerate code. So far as I can tell, only some source files and include directories are added to the makefile, but after this, whenever CLEAN REBUILD is selected, it gets a few lines in on the terminal and then returns the error:

make: *** No rule to make target `build/debug/Core/Src/stm32g4xx_it.o', needed by `build/maketest.elf'.  Stop.

jortbmd, 2024-12-23:
Hi thanks for opening up an issue for this. This should not happen and it might be an issue with cubeMX but I am not quite sure. Could you maybe share a minimum example of this where it happens so I can track down the issue?

jortbmd, 2025-01-20:
Closing this issue due to inactivity. Should the issue persist please feel free to re-open the issue.