bmd-studio/stm32-for-vscode #216: Could not find NodeJS link for win32, arm64 device

2BlackCats, 2024-11-14:
```Something has gone wrong while installing the build tools: Error: Could not find the NodeJS link for your specific platform: platform: win32, arch: arm64. Please open a GitHub Issue: ```

2BlackCats, 2024-11-15:
Was able to manually add openocd, and make from chocolatey to get the program to work.

jortbmd, 2024-11-18:
Thanks for opening up an issue! Ah yes there is no default support for arm64 in the version of node currently used. I will put it on the to do list for updates to check if newer version do support it and maybe migrate to that specific node version. For now your fix is perfect and should work just fine with STM32 for VSCode.

jortbmd, 2025-01-20:
The issue should be resolved with version v3.2.10. For this reason I am closing this issue. Should it still persist or if a new issue pops up please do not hesitate to re-open this issue or create a new one.

closed 2025-01-20 by jortbmd