bmd-studio/stm32-for-vscode #218: No Link found for this specific platform

gupengzu, 2024-11-26:
when I click the "install build tools",Something has gone wrong while installing the build tools: Error: No link found for this specific platform. Please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issue.

gupengzu, 2024-11-26:
[image]

gupengzu, 2024-11-26:
Please help me ,I have tried for one week!!!

jortbmd, 2024-11-26:
Hi. Thanks for opening up an issue for this. Could you maybe tell me which platform you are on? And could it maybe be a similar issue as: https://github.com/bmd-studio/stm32-for-vscode/issues/216 ?

gupengzu, 2024-11-28:
vscode

gupengzu, 2024-11-28:
> My apologies should have been more specific. As in do you use Windows, Linux or Mac and which architecture Intelx86 Apple ARM etc.

thank you for your care. I have download gcc and add its path to the environment.
So,i have enter the interface.But,when I click the "build ",it said:"build have some wrongs ",but it didn't tell me what wrongs them are.

2037920705, 2024-11-28:
I've also run into the same issue.

char990, 2024-12-03:
I have same issue.
Windows 10 Pro 22H2.
VSCode Version: 1.95.3

phann21, 2024-12-06:
I am seeing the same issue with some members of my team. We are all using x86 architecture.

rks6004, 2024-12-06:
Yep, can chime in here as a member of @phann21's team: will add my specifications here for reference:
Intel Core Ultra 7 155H (Meteor Lake-H)
Windows 11
We have tried alternating between the current version (3.2.8) and the previous 3.2.6 to see if the problem is version-specific, to no avail.

jortbmd, 2025-01-20:
Hi all. This issue should be resolved for version v3.2.10 if for this reason I am closing this issue for now. If this issue persists please re-open this issue or create a new one.

fnndyl, 2025-01-31:
Hi everyone, seeing if I can reopen this issue as I am having a similar proble, though in stranger circumstance. I installed everything without too much issue on my work MacBook Pro, but I needed to make a new account on the laptop, and trying to install it on the second account is giving the no link error again:

```Something has gone wrong while installing the build tools: Error: No link found for this specific platform. Please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issues```

```An error occurred during the node installation process. Please try again, or create an issue at: https://github.com/bmd-studio/stm32-for-vscode/issues. Include the error: Error: No link found for this specific platform. Please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issues```

Any idea why this would only be happening on the second account? Very odd. Happy to update with any OS/device info needed.

jortbmd, 2025-01-31:
Hi!
Thanks for responding to this issue. This should be fixed in the latest version. Could you maybe check if you have the latest version of the extension installed?
Also what type of macbook are you on the apple silicon (m1 and up) or intel silicon?

0-haha, 2025-02-01:
> Hi! Thanks for responding to this issue. This should be fixed in the latest version. Could you maybe check if you have the latest version of the extension installed? Also what type of macbook are you on the apple silicon (m1 and up) or intel silicon?

## Solution

If you change 

https://github.com/bmd-studio/stm32-for-vscode/blob/62524f64f6dd8cbce8470e11a90ee75e2cde78cb/src/buildTools/installTools.ts#L211

to `const nodeLatestURL = 'https://nodejs.org/dist/v16.20.2/'; ` it works.

Or change all regexs like `/href="(\/dist\/v\d+\.\d+\.\d+\/node-v\d+\.\d+\.\d+-win-x86.zip)/gm` to `/href="(\/dist\/.+\/node-v\d+\.\d+\.\d+-win-x86.zip)/gm,` in
https://github.com/bmd-studio/stm32-for-vscode/blob/62524f64f6dd8cbce8470e11a90ee75e2cde78cb/src/buildTools/installTools.ts#L157-L175

The extension uses node 16 which is not outdated. Not sure if it is possible to pump the node version or make the node version as a variable in the setting so that users can change it

## Why
This is a problem of node repo. Here is why.

`installTools.ts` pulls node version from
https://github.com/bmd-studio/stm32-for-vscode/blob/62524f64f6dd8cbce8470e11a90ee75e2cde78cb/src/buildTools/installTools.ts#L211

and then extract the right version in the function getPlatformSpecificNodeLink
https://github.com/bmd-studio/stm32-for-vscode/blob/62524f64f6dd8cbce8470e11a90ee75e2cde78cb/src/buildTools/installTools.ts#L184-L208

The regex pattern is nodeRegex defined as
https://github.com/bmd-studio/stm32-for-vscode/blob/62524f64f6dd8cbce8470e11a90ee75e2cde78cb/src/buildTools/installTools.ts#L157-L175

However, the response of `https://nodejs.org/dist/latest-gallium/` is
```
<!DOCTYPE html><html>
<head>
  <title>Index of /dist/latest-gallium/</title>
  <style>   
    @media (prefers-color-scheme: dark) {
      body {
        color: white;
        background-color: #1c1b22;
      }
      a {
        color: #3391ff;
      }
      a:visited {
        color: #C63B65;
      }
    }
  </style>
</head>
<body>
<h1>Index of /dist/latest-gallium/</h1><hr><pre><a href="../">../</a>
<a href="docs/">docs/</a>                                                             -                   -
<a href="win-x64/">win-x64/</a>                                                          -                   -
<a href="win-x86/">win-x86/</a>                                                          -                   -
<a href="/dist/latest-gallium/SHASUMS256.txt.asc">SHASUMS256.txt.asc</a>                                 04-Nov-2024 17:25               3.9 KB
<a href="/dist/latest-gallium/SHASUMS256.txt.sig">SHASUMS256.txt.sig</a>                                 04-Nov-2024 17:25                438 B
<a href="/dist/latest-gallium/SHASUMS256.txt">SHASUMS256.txt</a>                                     04-Nov-2024 17:25               3.2 KB
<a href="/dist/latest-gallium/node-v16.20.2-aix-ppc64.tar.gz">node-v16.20.2-aix-ppc64.tar.gz</a>                     30-Oct-2024 17:56                45 MB
<a href="/dist/latest-gallium/node-v16.20.2-darwin-arm64.tar.gz">node-v16.20.2-darwin-arm64.tar.gz</a>                  30-Oct-2024 17:56                30 MB
<a href="/dist/latest-gallium/node-v16.20.2-darwin-arm64.tar.xz">node-v16.20.2-darwin-arm64.tar.xz</a>                  30-Oct-2024 17:56                19 MB
<a href="/dist/latest-gallium/node-v16.20.2-darwin-x64.tar.gz">node-v16.20.2-darwin-x64.tar.gz</a>                    30-Oct-2024 17:56                31 MB
<a href="/dist/latest-gallium/node-v16.20.2-darwin-x64.tar.xz">node-v16.20.2-darwin-x64.tar.xz</a>                    30-Oct-2024 17:56                21 MB
<a href="/dist/latest-gallium/node-v16.20.2-headers.tar.gz">node-v16.20.2-headers.tar.gz</a>                       04-Nov-2024 17:25               569 KB
<a href="/dist/latest-gallium/node-v16.20.2-headers.tar.xz">node-v16.20.2-headers.tar.xz</a>                       04-Nov-2024 17:25               386 KB
<a href="/dist/latest-gallium/node-v16.20.2-linux-arm64.tar.gz">node-v16.20.2-linux-arm64.tar.gz</a>                   30-Oct-2024 17:56                34 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-arm64.tar.xz">node-v16.20.2-linux-arm64.tar.xz</a>                   30-Oct-2024 17:56                22 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-armv7l.tar.gz">node-v16.20.2-linux-armv7l.tar.gz</a>                  30-Oct-2024 17:56                31 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-armv7l.tar.xz">node-v16.20.2-linux-armv7l.tar.xz</a>                  30-Oct-2024 17:56                19 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-ppc64le.tar.gz">node-v16.20.2-linux-ppc64le.tar.gz</a>                 30-Oct-2024 17:56                36 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-ppc64le.tar.xz">node-v16.20.2-linux-ppc64le.tar.xz</a>                 30-Oct-2024 17:56                23 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-s390x.tar.gz">node-v16.20.2-linux-s390x.tar.gz</a>                   30-Oct-2024 17:56                34 MB
<a href="/dist/latest-gallium/node-v16.20.2-linux-s390x.tar.xz">node-v16.20.2-linux-s390x.tar.xz</a>                   30-Oct-2024 17:56                22 MB
[... 48 lines cut ...]
<a href="/dist/v16.20.2/node-v16.20.2-headers.tar.gz">node-v16.20.2-headers.tar.gz</a>                       04-Nov-2024 17:25               569 KB
[... cut ...]
<a href="/dist/v16.20.2/node-v16.20.2.tar.gz">node-v16.20.2.tar.gz</a>                               30-Oct-2024 17:56                70 MB
<a href="/dist/v16.20.2/node-v16.20.2.tar.xz">node-v16.20.2.tar.xz</a>                               30-Oct-2024 17:56                37 MB
</pre><hr /></body>
</html>
```
In this case, on linux 64, the regex  `/href="(\/dist\/v\d+\.\d+\.\d+\/node-v\d+\.\d+\.\d+-linux-x64.tar.gz)/gm` cannot match `<a href="/dist/latest-gallium/node-v16.20.2-linux-x64.tar.xz">node-v16.20.2-linux-x64.tar.xz</a>`. Thus, link becomes undefined.

fnndyl, 2025-02-04:
Can confirm does not work on latest version of the VSCode extension. Is the above fix something I can implement? Or do I need to wait for the extension to be updated.

jortbmd, 2025-02-04:
Thank you for all the input. The newest version has been released and should resolve this issue. If you could check if it fixed the problem for you that would be greatly appreciated.

fnndyl, 2025-02-04:
Looks like it's fixed! Thank you @jortbmd and @0-haha that was quick.