bmd-studio/stm32-for-vscode #219: Install Build Tools Error: getaddrinfo ENOTFOUND nodejs.org

KristjanB, 2024-12-26:
I'm trying to install all the necessary packages using "Install Build Tools" but keep getting this error:

Something has gone wrong while installing the build tools: Error: getaddrinfo ENOTFOUND nodejs.org

I'm on windows 10, using Intel i7 processor.

Thanks!

KristjanB, 2024-12-26:
Acutally I got two more error messages:

`Something has gone wrong while installing the build tools: Error: No link found for this specific platform. Please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issues`

`An error occurred during the node installation process. Please try again, or create an issue at: https://github.com/bmd-studio/stm32-for-vscode/issues. Include the error: Error: No link found for this specific platform. Please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issues`

jortbmd, 2025-01-20:
Hi thanks for opening up an issue. I have taken the issue into account and it should be fixed for version v3.2.10. For this reason I am closing the current issue. Should any similar issue arise feel free to re-open this issue or create a new one.

closed 2025-01-20 by jortbmd