bmd-studio/stm32-for-vscode #220: make: *** No rule to make target 'build/debug/main.o'

DeepooBelief, 2025-01-16:
Hi, I tried using this extension, but encountered an error when attempting to switch the project to C++.  
Is there a step I might have missed? 
Thank you!  

## Environment Details
- **OS:** Windows 11 (version 24H2, x64)
- **Extension Version:** 3.2.10
- **Make Version:** 4.4.1

## Steps to Reproduce
1. Generate the project code using **CubeMX**.
2. Verify that the extension builds the project successfully in C mode.
3. Rename `main.c` to `main.cpp`.
   Update `STM32-for-VSCode.config.yaml` to set the project type to **C++**.
4. Click the **Build** button provided by the extension.

## Observed Behavior
The build fails with the following error:
`make: *** No rule to make target 'build/debug/main.o'`

## Temp solution
1. If I change the `CXX_SOURCES` in **STM32Make.make** to `CPP_SOURCES`, the make command can run successfully.
2. Use 3.2.6 instead.
[image]

wrs, 2025-01-19:
Same here, it's not building any of my .cpp files now, but fix (1) above works.

jortbmd, 2025-01-20:
Thanks for opening up and issue. I have addressed this in a new hotfix version 3.2.11, which should fix this issue. The new version is out now. Please let me know if the issue persists or not.

closed 2025-01-20 by DeepooBelief