bmd-studio/stm32-for-vscode #223: openocd.cfg file for STM32F3x boards (Error: error writing to flash at address 0x08000000 at offset 0x00000000)

grybouilli, 2025-04-28:
When trying to flash to a [legit] SMT32F303K8 using the STM32-for-vscode extension, I was encountering an issue similar to the following subject:
[Can’t load to Nucleo_f303k8](https://community.platformio.org/t/cant-load-to-nucleo-f303k8/34819)

The issue is that while trying to flash, the following error occurs:
```bash
[stm32f3x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000300 msp: 0x20003000
** Programming Started **
Warn : Adding extra erase range, 0x08007c30 … 0x08007fff
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:1516: Error: ** Programming Failed **
in procedure ‘program’
```
One of the solution offered in the attached topic gave the folllowing link : [https://techoverflow.net/2022/02/12/how-to-unlock-stm32f0x-using-openocd/](https://techoverflow.net/2022/02/12/how-to-unlock-stm32f0x-using-openocd/) in which it is stated that
"**OpenOCD uses the same backend for STM32F1x and STM32F0x.**"

So, I went ahead and changed the line :
```cfg
source [find target/stm32f3x.cfg]
```

for 
```cfg
source [find target/stm32f1x.cfg]
```

in the file `openocd.cfg` generated by the extension. With this, I was able to flash to the STM32F303K8 board.

I don't have other STM32F3x boards than the STM32F303K8, so I can't test for other references, but it could be nice to integrate that fix to the extension. I don't have experience developing vscode extensions, so I could look into making a pull request myself but that would take some time.

grybouilli, 2025-04-28:
Thanks for the fast reply.

I have tried updating, and have also tried removing and re-installing the build tools, the result is the same ; if `openocd.cfg` contains:
* `source [find target/stm32f3x.cfg]` : flashing will fail with error:
```bash
Error: Failed to write memory at 0x20003004
Error: error writing to flash at address 0x08000000 at offset 0x00000000
```
* `source [find target/stm32f1x.cfg]` : flashing will be successful

jortbmd, 2025-06-09:
Thanks for replying. It is a nice work around. But I am a bit hesitant to fix this in the extension as it is very much a bug with openOCD. Will leave it open for now to track the issue and have some visibility for others which experience the same issue.