bmd-studio/stm32-for-vscode #226: Character limit causes linker to fail

has213, 2025-05-13:
Was using an older version of stm32-for-vscode and have now gone to the newest 3.2.13. Have a big project and i thinge the following line in make should help with this but in my case the .in file ends after 8192 characters (like `... build/debug/FastMathFunctions.o buil`).

```make
$(RELEASE_DIRECTORY)/$(TARGET).elf: $(OBJECTS) STM32Make.make | $(RELEASE_DIRECTORY)
	@echo $(OBJECTS) > $@.in #this line
	$(CC) @$@.in $(LDFLAGS) -o $@
	$(SZ) $@
```

I am on windows and I thinke because of git for windows the bash is available. the flowing works for me and should also work on most other systems. for macOS im not sure because of the use of zsh.

```make
$(RELEASE_DIRECTORY)/$(TARGET).elf: $(OBJECTS) STM32Make.make | $(RELEASE_DIRECTORY)
	@bash -c 'echo $(OBJECTS) > $@.in' #this line
	$(CC) @$@.in $(LDFLAGS) -o $@
	$(SZ) $@
```

has213, 2025-05-13:
Additional note:
You can make the .elf.in file more readable with putting each object in a seprate line

```make
$(RELEASE_DIRECTORY)/$(TARGET).elf: $(OBJECTS) STM32Make.make $(RELEASE_DIRECTORY)/res.o $(LDSCRIPT) | $(RELEASE_DIRECTORY)
	@bash -c 'printf "%s\n" $(OBJECTS) > $@.in'
	$(CC) @$@.in $(RELEASE_DIRECTORY)/res.o $(LDFLAGS) -o $@
	$(SZ) $@
```

jortbmd, 2025-06-09:
Thanks for opening up an issue and suggesting a fix! The git bash fix might be nice, however I am a bit hesitant to put in more external dependencies in the project. It might be nice to have a selectable shell for execution either way.
For now I will look into if I can solve the issue using the Makefile syntax itself. Will keep you posted on this, because it is an issue I would very much like to solve.

jortbmd, 2025-06-10:
Was checking for a quick fix for you. Can you check if the following release would solve your issue?
https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.14-beta
You can install it by using the following command (use ctrl+shift+p to bring up the command menu): Extensions: Install from VSIX.